micro-ROS/micro-ROS-Agent #209: Topic Not Visible error

RaghuRamaBL, 2023-10-18:
## Issue template

- Hardware description: ESP32-WROVER-E
- RTOS:   ROS2 Humble
- Installation type: micro_ros_setup
- Version or commit hash: Ubuntu 22.04.3 LTS, Arduino 1.8.19
#### Steps to reproduce the issue
# Source the ROS 2 installation
source /opt/ros/$ROS_DISTRO/setup.bash

# Create a workspace and download the micro-ROS tools
mkdir microros_ws
cd microros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup

# Update dependencies using rosdep
sudo apt update && rosdep update
rosdep install --from-paths src --ignore-src -y

# Install pip
sudo apt-get install python3-pip

# Build micro-ROS tools and source them
colcon build
source install/local_setup.bash

# Create step
ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32

# Configure step with ping_pong app and serial transport
ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport serial

I followed all these steps no issues till now

# Build step
ros2 run micro_ros_setup build_firmware.sh

when i executed command in between it shows as "colcon build stderr output"
later it will complete the execution without any error 100% build target app.

# Flash step
ros2 run micro_ros_setup flash_firmware.sh

# Download micro-ROS-Agent packages
ros2 run micro_ros_setup create_agent_ws.sh

# Build step
ros2 run micro_ros_setup build_agent.sh

while executing this line it gives 

$ ros2 run micro_ros_setup build_agent.sh
source install/local_setup.bash
Building micro-ROS Agent
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [6.49s]                     
Starting >>> micro_ros_agent
[Processing: micro_ros_agent]                             
--- stderr: micro_ros_agent                                    
Cloning into 'xrceagent'...
Switched to a new branch 'ros2'
HEAD is now at 2cbbc4a Fix deserialization endianness (#336)
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< micro_ros_agent [53.1s]

Summary: 2 packages finished [1min 1s]
  1 package had stderr output: micro_ros_agent

source install/local_setup.bash
# Run a micro-ROS agent
$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_8484ebb7cd24ec11ad45dd31b659684c-if00-port0
executes correctly

[1697617695.989983] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1697617695.990224] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1697617697.966921] info     | Root.cpp           | create_client            | create                 | client_key: 0x78DB3FD8, session_id: 0x81
[1697617697.967054] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x78DB3FD8, address: 0
[1697617697.994969] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x78DB3FD8, participant_id: 0x000(1)
[1697617698.013419] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x78DB3FD8, topic_id: 0x000(2), participant_id: 0x000(1)
[1697617698.026173] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x78DB3FD8, publisher_id: 0x000(3), participant_id: 0x000(1)
[1697617698.039648] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x78DB3FD8, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1697617698.057389] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x78DB3FD8, topic_id: 0x001(2), participant_id: 0x000(1)
[1697617698.069736] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x78DB3FD8, publisher_id: 0x001(3), participant_id: 0x000(1)
[1697617698.083964] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x78DB3FD8, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[1697617698.101279] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x78DB3FD8, topic_id: 0x002(2), participant_id: 0x000(1)
[1697617698.114001] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x78DB3FD8, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1697617698.128575] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x78DB3FD8, datareader_id: 0x000(6), subscriber_id: 0x000(4)
[1697617698.148276] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x78DB3FD8, topic_id: 0x003(2), participant_id: 0x000(1)
[1697617698.160762] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x78DB3FD8, subscriber_id: 0x001(4), participant_id: 0x000(1)
[1697617698.175174] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x78DB3FD8, datareader_id: 0x001(6), subscriber_id: 0x001(4)

In another terminal if i run 
source /opt/ros/$ROS_DISTRO/setup.bash

# Subscribe to micro-ROS ping topic
ros2 topic echo /microROS/ping

 ros2 topic echo /microROS/ping
WARNING: topic [/microROS/ping] does not appear to be published yet
Could not determine the type for the passed topic

I have gone through Domail ID and set it to even 0 then also i am getting the same result.
I even refered this and other links regarding the same https://github.com/micro-ROS/micro_ros_arduino/issues/21

This line also gets exicuited correctly code gets uploaded to the board.

#### Expected behavior
You should see the topic messages published by the Ping Pong node every 5 seconds:

user@user:~$ ros2 topic echo /microROS/ping
stamp:
  sec: 20
  nanosec: 867000000
frame_id: '1344887256_1085377743'
---
stamp:
  sec: 25
  nanosec: 942000000
frame_id: '730417256_1085377743'
ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
#### Actual behavior
 WARNING: topic [/microROS/ping] does not appear to be published yet
Could not determine the type for the passed topic
#### Additional information

pablogs9, 2023-10-19:
Could you share you micro-ROS Agent output with the flag `-v6` ?

RaghuRamaBL, 2023-10-21:
raghurama@raghurama-Latitude-5490:~/microros_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_8484ebb7cd24ec11ad45dd31b659684c-if00-port0 -v6
[1697868513.201535] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1697868513.201753] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1697868527.599355] info     | Root.cpp           | create_client            | create                 | client_key: 0x796283A4, session_id: 0x81
[1697868527.599396] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x796283A4, address: 0
[1697868527.599522] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x796283A4, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1697868527.612081] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 44, data: 
0000: 81 80 00 00 01 07 24 00 00 0A 00 01 01 03 00 00 16 00 00 00 00 01 A5 A5 0E 00 00 00 70 69 6E 67
0020: 70 6F 6E 67 5F 6E 6F 64 65 00 00 00
[1697868527.637552] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x796283A4, participant_id: 0x000(1)
[1697868527.637670] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x796283A4, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1697868527.637708] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x796283A4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1697868527.649675] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1697868527.655242] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 80, data: 
0000: 81 80 01 00 01 07 47 00 00 0B 00 02 02 03 00 00 39 00 00 00 11 00 00 00 72 74 2F 6D 69 63 72 6F
0020: 52 4F 53 2F 70 69 6E 67 00 00 01 00 1D 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64
0040: 64 73 5F 3A 3A 48 65 61 64 65 72 5F 00 00 01 00
[1697868527.655354] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x796283A4, topic_id: 0x000(2), participant_id: 0x000(1)
[1697868527.655432] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x796283A4, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1697868527.655450] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x796283A4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1697868527.667195] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1697868527.667222] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 24, data: 
0000: 81 80 02 00 01 07 10 00 00 0C 00 03 03 03 00 00 02 00 00 00 00 00 00 01
[1697868527.667300] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x796283A4, publisher_id: 0x000(3), participant_id: 0x000(1)
[1697868527.667362] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x796283A4, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 03 00 00
[1697868527.667380] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x796283A4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1697868527.680036] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1697868527.680065] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 36, data: 
0000: 81 80 03 00 01 07 1C 00 00 0D 00 05 05 03 00 00 0E 00 00 00 00 02 01 00 03 00 01 3F 0A 00 00 00
0020: 00 00 00 03
[1697868527.680385] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x796283A4, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1697868527.680436] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x796283A4, len: 14, data: 
0000: 81 80 03 00 05 01 06 00 00 0D 00 05 00 00
[1697868527.680455] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x796283A4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1697868527.691722] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 04 00 00 00 80
[1697868527.697270] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 80, data: 
0000: 81 80 04 00 01 07 47 00 00 0E 00 12 02 03 00 00 39 00 00 00 11 00 00 00 72 74 2F 6D 69 63 72 6F
0020: 52 4F 53 2F 70 6F 6E 67 00 00 01 3F 1D 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64
[... cut ...]
[1697868527.733681] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 07 00 00 00 80
[1697868527.739241] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x796283A4, len: 80, data: 
0000: 81 80 07 00 01 07 47 00 00 11 00 22 02 03 00 00 39 00 00 00 11 00 00 00 72 74 2F 6D 69 63 72 6F
0020: 52 4F 53 2F 70 69 6E 67 00 00 01 00 1D 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64
0040: 64 73 5F 3A 3A 48 65 61 64 65 72 5F 00 00 01 00

This was the result

RaghuRamaBL, 2023-10-21:
one doubt may i know what is the significance of the -v6 flag i am new to ros

pablogs9, 2023-10-24:
It is setting a verbosity to max level. Is this the whole log or there are more lines?

RaghuRamaBL, 2023-10-24:
> It is setting a verbosity to max level. Is this the whole log or there are more lines?

It was repeating there were more line

pablogs9, 2023-10-24:
In any case, the way to go when using ESP32-based boards is the micro-ROS Component for IDF: https://github.com/micro-ROS/micro_ros_espidf_component

RaghuRamaBL, 2023-10-24:
> In any case, the way to go when using ESP32-based boards is the micro-ROS Component for IDF: https://github.com/micro-ROS/micro_ros_espidf_component
> 
> Could you give it a try and report if the same error happens?

Hi thanks for the reply
I am new to ros can i get any steps to proceed so that i can try using repository that you have suggested
Should i follow same as what i was doing previously but in the place if micro_ros_setup do i need to change it to micro_ros_espidf_component ?