micro-ROS/micro-ROS-Agent #210: Unable to remap topics

adityakamath, 2023-10-22:
I have a micro-ROS node that is subscribing to a /cmd_vel topic. I am currently launching the node like the following from a python launch file:

```
        Node(
            package='micro_ros_agent',
            executable='micro_ros_agent',
            name='micro_ros_agent',
            output='screen',
            arguments=['serial', '--dev', '/dev/ttyUSB0'],
            remappings=[
                ('/cmd_vel', '/joy_vel'),
            ]),
```

The node runs as expected, but it still subscribes to the /cmd_vel topic, and not the /joy_vel topic. I am using ROS2 Humble and the latest micro_ros_agent

pablogs9, 2023-10-24:
Hello, there is not support for topic remapping at micro-ROS Agent level.

closed 2023-10-24 by pablogs9

adityakamath, 2023-10-28:
Thanks for the update, its a shame, it would be really useful to have topic remapping at the micro-ROS agent level..

pablogs9, 2023-10-28:
Sure, PRs are welcomed, please let us know if you plan to contribute.

adityakamath, 2023-12-02:
@pablogs9 would love to contribute, I need something to work on during my winter break anyway. But I am not entirely sure where to start from, any pointers would be helpful!