micro-ROS/micro-ROS-Agent #211: No response to UDP connection request

luukWeFab, 2023-11-01:
## Issue template

- Hardware description: STM32F446VET
- RTOS: FreeRTOS
- Installation type: [micro_ros_stm32cubemx_utils](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils)
- Version or commit hash: humble

#### Steps to reproduce the issue
1. Have a serial connection established using the custom transport layer specified in the [micro_ros_stm32cubemx_utils](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils) repository.
2. Change the custom transport layer to work through a w5500 chip. (Using the [ioLibrary_Driver](https://github.com/Wiznet/ioLibrary_Driver) library)
`#ifndef __W5500_TRANSPORT_LAYER_HPP
#define __W5500_TRANSPORT_LAYER_HPP

#include <uxr/client/transport.h>

#include "socket.h"
#include "dhcp.h"
#include "dns.h"
#include "spi.h"
#include "rcl_error_handler.hpp"
#include "pinout.hpp"

#include "cmsis_os.h"

#define DHCP_SOCKET     0
#define DNS_SOCKET      1
#define HTTP_SOCKET     2
#define MICRO_ROS_SOCKET 3

/**
 * @brief The socket of the w5500 chip that will connect to microROS
 * 
 */
extern uint8_t microROS_agent_socket;

extern uint8_t rx_buffer[cnst::RX_SOCKET_BUFFER];
extern uint32_t curr_head;
extern uint32_t curr_tail;
extern volatile uint32_t curr_length;

template<uint32_t Size>
struct dma_tx_buffer_t {
    uint8_t buffer[Size];
    bool transmission_started;
};

/**
 * @brief Select the spi of the w5500
 */
extern void W5500_Select(void);

/**
 * @brief Unselect the spi of the w5500
 */
extern void W5500_Unselect(void);

/**
 * @brief Read the spi buffer
 * 
 * @param buff (uint8_t *): the buffer to store the data in
 * @param len (uint16_t): how many bytes of data to read
 */
extern void W5500_ReadBuff(uint8_t* buff, uint16_t len);

/**
 * @brief Write to the spi buffer
 * 
 * @param buff (uint8_t *): the buffer which data should be writte
 * @param len (uint16_t): how many bytes of data to write.
 */
[... 269 lines cut ...]
    }
}
`
3. Start the micro-ros-agent with udp4, on port 8888.

#### Expected behavior
microROS agent logging the recv_message, and responding with a send_message that it connects.

#### Actual behavior
The microROS agent logs the recv_message, however it doesn't respond to it. 
[image]

#### Additional information
- I checked that what the microROS client tries to send through the w5500_transport_write function is the same as what the microROS agent is printing, so the flow from microROS client to microROS agent should be correct.
- I compared what the microROS client tries to send through the w5500_transport_write function, and the original custom_transport_write function, and both seem similar.
Serial (uart) communication:
[image]
SPI (w5500) communication:
[image]
Received by microROS agent over UDP:
[image]

- I tried to log what the microROS agent is receiving from the serial communication with tag -v6, however it is not logging its reception.

Basically my question is why is my client not connecting to the microROS agent over udp, but is through serial while the request seems to be similar?

pablogs9, 2023-11-02:
Serial and Networked transport are not handled in the same way by the library, check serial framing here: https://micro-xrce-dds.docs.eprosima.com/en/latest/transport.html#stream-framing-protocol

Are you taking into account that framing protocol shall be disabled when using UDP by means of this flag: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/blob/3819f6819cee1ff01c3da58ecbc8bfcbf4b0a2a5/sample_main.c#L382

luukWeFab, 2023-11-02:
I did not accounting that the framing protocol needed to be disabled thank you. Changing 
[image]
to 
[image]
indeed made the udp agent respond to the request. Thank you.

A second quick question I have is whether TCP is supported or not? Because that would then also work without framing enabled right? I ask this because I read in the main readme of this repository that it is, however in the readme of [the micro_ros_agent sub folder](https://github.com/micro-ROS/micro-ROS-Agent/blob/iron/micro_ros_agent/README.md) it is stated that only UDP, and serial are supported.

And thank you again for the help!

pablogs9, 2023-11-02:
Nice to hear that @luukWeFab, I'm closing as solved.

TCP is supported at XRCE-DDS level but it not very well integrated in micro-ROS. 

Take a look here to a sample implementation: https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/master/src/c/profile/transport/ip/tcp/tcp_transport_posix.c

As you can see, there is another "framing" for TCP, it is handled at XRCE Client level here: https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/97175304425c5bee87c6fddd99de1ef8d0c394dc/src/c/profile/transport/ip/tcp/tcp_transport.c#L33

closed 2023-11-02 by pablogs9