micro-ROS/micro-ROS-Agent #215: micro-ROS-Agent Serial Only Works for 115200 BAUD

Jake-Carter, 2024-01-19:
**Describe the bug**

The micro-ROS agent only seems to work with a serial connection at 115200 baud.  Attempting to connect at any other baudrate results in no connection.

```shell
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 --baud 230400
```

Even at `-v6` there are no error messages, and I can see serial activity coming into the right port as my device attempts to connect.  It has the symptoms of a baudrate mismatch.

```shell
[1705630853.830796] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1705630853.830968] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```

**To Reproduce**
Steps to reproduce the behaviour:
1. Attempt to connect to the agent over a serial port at any BAUD rate except 115200

**System information (please complete the following information):**
 - OS: Ubuntu 22.04 LTS
 - ROS 2 Humble
 - Version [humble](https://github.com/micro-ROS/micro-ROS-Agent/tree/humble) (I believe - installed via micro_ros_setup_tools)

**Additional context**

I'm currently working on a custom platform, and don't have any other hardware to test.  I've ordered an ST Nucleo board to have another reference, but if anyone can run a test on a known working platform I would greatly appreciate it.

pablogs9, 2024-01-19:
Is possible to open your serial device with other baudrate for other purposes?

I mean, if you open a miniterm/minicom with a different baudrate, do you receive data from the micro-ROS Client?

Jake-Carter, 2024-01-19:
Here is the miniterm output at 115200 baud:

```serial
~ » python -m serial.tools.miniterm /dev/ttyACM0 115200                                               jhcarter@jhcarter-X670E
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

-=- MAX78000 micro-ROS + FreeRTOS (V10.5.1) Demo -=-
SystemCoreClock = 60000000
Initializing RTC
Assigning custom transports
Starting scheduler in 1s...
~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏t�,-�␀�␇S,~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏t�,-�␀�␇S,~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏t�,-�␀�␇S,~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏t�,-�␀�␇S,~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏t�,-�␀�␇S,~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏t�,-�␀�␇S,~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏t�,-�␀�␇S,
--- exit ---
```

and at 230400 baud

```serial
~ » python -m serial.tools.miniterm /dev/ttyACM0 230400                                               jhcarter@jhcarter-X670E
--- Miniterm on /dev/ttyACM0  230400,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

-=- MAX78000 micro-ROS + FreeRTOS (V10.5.1) Demo -=-
SystemCoreClock = 60000000
Initializing RTC
Assigning custom transports
Starting scheduler in 1s...
�>V�␀�␇0�~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏
--- exit ---
```

So it seems my `printf` statements are working normally at the different baud rates, but not my custom serial transport functions.  I will check my implementation.  Closing as this is most likely an issue on my side.

Jake-Carter, 2024-01-20:
Hi @pablogs9, re-opening as I fixed the issues with my serial transports but the issues still persists.

I set the client key in my application code so that it would not randomly generate one each time.  At 57600, 115200, and 230400 I logged the miniterm output to a file, and a diff shows they are all identical.
```shell
~/repos/msdk/Examples/MAX78000/micro-ROS/object_detector (dev/micro-ros*) » python -m serial.tools.miniterm /dev/ttyUSB0 230400 | tee log_230400
--- Miniterm on /dev/ttyUSB0  230400,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

-=- MAX78000 micro-ROS + FreeRTOS (V10.5.1) Demo -=-
SystemCoreClock = 60000000
Initializing RTC
Testing all ASCII characters...
---
␀␁␂␃␄␅␆␇
␎␏␐␑␒␓␔␕␖␗␘␙␚␛␜␝␞␟ !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~␡��������������������������������������������������������������������������������������������������������������������������������
---
Assigning custom transports
Starting scheduler in 1s...
~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏�^�␑�␀�␇␐  ~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏�^�␑�␀�␇␐  ~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏�^�␑�␀�␇␐  ~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏�^�␑�␀�␇␐~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏�^�␑�␀�␇␐   ~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏�^�␑�␀�␇␐  ~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏�^�␑�␀�␇␐  ~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏�^�␑�␀�␇␐~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏�^�␑�␀�␇␐   ~␀␀␘␀�␀␀␀␀␁␐␀XRCE␁␀␁␏�^�␑�␀�␇␐
--- exit ---
```

[log_57600.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/13995975/log_57600.txt)
[log_115200.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/13995976/log_115200.txt)
[log_230400.txt](https://github.com/micro-ROS/micro-ROS-Agent/files/13995977/log_230400.txt)

However, the agent only establishes the session at 115200...  The others hang with no error messages.

```shell
~/repos/msdk/Examples/MAX78000/micro-ROS/object_detector (dev/micro-ros*) » ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baud 115200
[1705710566.747570] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1705710566.747742] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1705710583.128053] info     | Root.cpp           | create_client            | create                 | client_key: 0xBA5EBA11, session_id: 0x81
[1705710583.128103] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0xBA5EBA11, address: 0
```

Do you have a platform to test at a non-default baud rate?

pablogs9, 2024-01-20:
Are you sharing the same UART for XRCE and printf? That would likely fail

Jake-Carter, 2024-01-23:
I am, but only call printf as logging info before running the appMain task.  

I removed all printfs and reassigned the stdio drivers to use a separate UART, but still have the same results

pablogs9, 2024-01-23:
Try using `-b` or `--baudrate`:

[image]

https://micro-xrce-dds.docs.eprosima.com/en/latest/agent.html?highlight=baud

Jake-Carter, 2024-01-23:
Jeez, yep.  Dumb mistake...  I was using `--baud` instead of `--baudrate`...  thanks.

closed 2024-01-23 by Jake-Carter