micro-ROS/micro-ROS-Agent #219: terminate called after throwing an instance of 'std::length_error'; what():  cannot create std::vector larger than max_size()

Johannes-42, 2024-02-07:
**Describe the bug**
I used microROS on my older Laptop, there it worked fine but now i switched to my newer PC and the microROS Agent crashes.

**To Reproduce**
I run it with Zephyr and freeRTOS. I used the Tutorials from the official microROS webpage.
https://micro.ros.org/docs/tutorials/core/first_application_rtos/zephyr/
(and equivalent freeRTOS tutorial)
This means i used the ping_pong example.

When running the command: ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/serial/by-id/usb-ZEPHYR_Zephyr_micro-ROS_203030335841500F-if00

i get the following output:
[1707298293.366673] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1707298293.366786] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1707298294.446095] info     | Root.cpp           | create_client            | create                 | client_key: 0x1D96764C, session_id: 0x81
[1707298294.446130] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1D96764C, address: 0
terminate called after throwing an instance of 'std::length_error'
  what():  cannot create std::vector larger than max_size()
[ros2run]: Aborted

**Expected behaviour**
Not crashing

**System information (please complete the following information):**
 - OS: Ubuntu 22.04.3 LTS
 - ROS 2 humble
 - Olimex STM32 Boars like in the Tutorial
 
 **Additional context**
It is the same for freeRTOS and Zephyr so it could be something with newer hardware. I also tried it with other examples but it will also crash.

pablogs9, 2024-02-07:
Are you mixing humble and iron distros?

Johannes-42, 2024-02-07:
I used ROS1 Noetic before on Ubuntu 20.04. Then i deleted that and upgraded to Ubunutu 22.04 and installed ROS2 humble. 
[image]

pablogs9, 2024-02-07:
Could you check that you micro-ROS agent is in humble also?

Johannes-42, 2024-02-07:
I do not know how to do that. 
But i just removed ROS2 and my microROS folder and reinstalled ROS2. Now it is working. 
I probably set it (ROS2) up wrong somewhere.

Thanks for the help!