micro-ROS/micro-ROS-Agent #221: micro-ROS-Agent on ARM32v7 (Odroid XU4)

elgarbe, 2024-03-03:
Hi, I need to use micro-ros-agent on docker on an ARM32v7. I build a docker image with ros2 humble from source, using it as base image, do I need to just follow the steps for build micro-ros-agent? (https://micro.ros.org/docs/tutorials/core/first_application_linux/) in a new dockerfile? don't you provide such an image, right?

pablogs9, 2024-03-04:
No we don't, the approach shall be to build the agent in the required platform

elgarbe, 2024-03-04:
Ok, I managed to make an image with micro-ros-agent. It doesn't work as expected. I mean, it works but once connected to my custom board (stm32f777) I get this:

```
[35m[1709577566.071391][m [32minfo    [m | [34mTermiosAgentLinux.cpp[m | [37minit                    [m | [32mrunning...[m             | fd: 3
[35m[1709577566.073126][m [32minfo    [m | [34mRoot.cpp          [m | [37mset_verbose_level       [m | [32mlogger setup[m           | verbose_level: 6
[35m[1709577566.123838][m [32minfo    [m | [34mRoot.cpp          [m | [37mcreate_client           [m | [32mcreate[m                 | client_key: 0x7A6BA362, session_id: 0x81
[35m[1709577566.123967][m [32minfo    [m | [34mSessionManager.hpp[m | [37mestablish_session       [m | [32msession established[m    | client_key: 0x7A6BA362, address: 0
[35m[1709577566.124302][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x7A6BA362, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[35m[1709577566.224730][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37mrecv_message            [m | [33m[==>> SER <<==][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[35m[1709577566.225230][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[35m[1709577566.325752][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37mrecv_message            [m | [33m[==>> SER <<==][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[35m[1709577566.326372][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[35m[1709577566.427721][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37mrecv_message            [m | [33m[==>> SER <<==][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[35m[1709577566.428263][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[35m[1709577566.528770][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37mrecv_message            [m | [33m[==>> SER <<==][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[35m[1709577566.529350][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[35m[1709577566.630760][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37mrecv_message            [m | [33m[==>> SER <<==][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[35m[1709577566.631252][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[35m[1709577566.731791][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37mrecv_message            [m | [33m[==>> SER <<==][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[35m[1709577566.732286][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[35m[1709577566.832684][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37mrecv_message            [m | [33m[==>> SER <<==][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[35m[1709577566.833386][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[35m[1709577566.934704][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37mrecv_message            [m | [33m[==>> SER <<==][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[35m[1709577566.935216][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[35m[1709577567.035780][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37mrecv_message            [m | [33m[==>> SER <<==][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[35m[1709577567.036370][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x7A6BA362, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
[35m[1709577583.918916][m [32minfo    [m | [34mTermiosAgentLinux.cpp[m | [37mfini                    [m | [32mserver stopped[m         | fd: 3
[... cut ...]
[35m[1709577586.405856][m [36mdebug   [m | [34mSerialAgentLinux.cpp[m | [37msend_message            [m | [33m[** <<SER>> **][m        | client_key: 0x687E95F7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 00 00 00 01 80
^C[ros2run]: Interrupt
```
On 1709577585.490859 I reset my board and then the connection is established again. In my board some object fail when I try to create them.
Is there an easy way to debug the problem with this messages?
BTW: In my embedded system I create a node two topics and two publishers.

elgarbe, 2024-03-04:
The problem is trying to create the node. 
I'm debuging my firmware, but some times the node is created and then micro-ros-agent gives:
[image]

I'm not sure what am I missing...

elgarbe, 2024-03-04:
Ok, I isolate the problem.
I moved from USB to UART in order to simplify communications.
I moved from my source build in docker on Odroid XU4 to official docker image on my laptop.
I use humble version of microros/micro_ros_static_library_builder in my embedded system.
So I was trying with humble docker image getting "Not enough memory error" then I've tried with iron version and everything works just fine:

pablogs9, 2024-03-05:
You need to have the same ROS 2 distro in all parts of your system. Humble and Iron may not be intercompatible.

elgarbe, 2024-03-05:
That's right, but I have used humble for the static library and humble for the agent and it doesn't work. The changed to iron agent and it started working.

pablogs9, 2024-03-05:
Is there any Iron node in your ROS 2 dataspace while you are running micro-ROS Humble?

elgarbe, 2024-03-05:
> Is there any Iron node in your ROS 2 dataspace while you are running micro-ROS Humble?

No, I just trying to get micros-ros working right now.

pablogs9, 2024-03-05:
So in that case, I need a replicator to have the same error on my local setup.

elgarbe, 2024-03-05:
I have a custom board based on stm32f777. I'm not sure how you can replicate it.
If humble library with humble agent are working ok then I will try to find the problem here. I can try starting with an empty system installing just humble.

BTW USB transport define a 2048 buffer size. I have to increase to 4096 in order to get it working. I was having issue on the node creation function, after that change using USB is working just fine.