micro-ROS/micro-ROS-Agent #222: Missing debian release for micro-ros-agent

pulak-gautam, 2024-03-05:
## Issue template

- Installation type: debian
- Version or commit hash: humble

#### Additional information

I was wondering why ``micro_ros_agent`` release repository was archived. This hinders us to make Debian releases of the packages  which have build-time dependency on ``micro_ros_agent``. Could you please clarify what steps we could take to circumvent around this?
Thank you

pablogs9, 2024-03-05:
`micro_ros_agent` is released as a Debian package in Vulcanexus: https://docs.vulcanexus.org/en/latest/rst/installation/linux_binary_installation.html

pulak-gautam, 2024-03-05:
Thank you. Will check out the same and will let you know if I face any hiccups.

closed 2024-03-05 by pulak-gautam