micro-ROS/micro-ROS-Agent #232: instruction for solving issue with this error : "   AttributeError: module 'em' has no attribute 'BUFFERED_OPT'  when building firmware for esp32 on udp transfering

AmirSamanMirjalili, 2024-08-18:
## Issue template

- Hardware description: esp32 
- RTOS:  freeRtos 
- Installation type: micro_ros_setup
- Version or commit hash: version of micro-ROS used: galactic for Ubuntu 20.04

#### Steps to reproduce the issue
Assuming you have already setup the firmware using `ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32` and after proper wifi configuration, when you try to build the firmware for udp transferring using `ros2 run micro_ros_setup build_firmware.sh` you will most likely get an error which related to the Python environment and the em (Empy) module. The specific error is: 
```
   AttributeError: module 'em' has no attribute 'BUFFERED_OPT'
```

#### To resolve this issue, follow these steps: 

1. First, deactivate any active Python virtual environments:

```
   deactivate
```

To install empy in the specific Python environment used by the build process, you should use the pip associated with that Python installation. Here's the command to install empy version 3.3.4 in that environment:
```
~/microros_ws/firmware/toolchain/espressif/python_env/idf4.1_py3.8_env/bin/pip3 install empy==3.3.4
```
2. After that, verify the installation:
```
~/microros_ws/firmware/toolchain/espressif/python_env/idf4.1_py3.8_env/bin/python3 -c "import em; print(em.__file__)"
```

3. Update setuptools and wheel:
```
~/microros_ws/firmware/toolchain/espressif/python_env/idf4.1_py3.8_env/bin/pip3 install --upgrade setuptools wheel
```

5. Clean your micro-ROS workspace:

```bash
cd ~/microros_ws
rm -rf build install log
```

6. Rebuild the workspace:

```bash
colcon build --symlink-install
```

7. Source the setup file:

```bash
source install/local_setup.bash
```

8. Try building the firmware again:

```bash
ros2 run micro_ros_setup build_firmware.sh
```

Highnoon0304, 2024-12-31:
안되는데요 씨2발련아?

qq1018408006, 2025-01-03:
> ## 问题模板
> * 硬件描述：esp32
> * 实时操作系统: freeRtos
> * 安装类型：micro_ros_setup
> * 版本或提交哈希：使用的微型 ROS 版本：Ubuntu 20.04 的 galactic
> 
> #### 重现问题的步骤
> 假设您已经使用 设置了固件，`ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32`并且正确配置了 wifi，当您尝试使用 构建固件以进行 udp 传输时，`ros2 run micro_ros_setup build_firmware.sh`您很可能会收到与 Python 环境和 em (Empy) 模块相关的错误。具体错误是：
> 
> ```
>    AttributeError: module 'em' has no attribute 'BUFFERED_OPT'
> ```
> 
> #### 若要解决此问题，请按照下列步骤操作：
> 1. 首先，停用所有活动的 Python 虚拟环境：
> 
> ```
>    deactivate
> ```
> 
> 要在构建过程使用的特定 Python 环境中安装 empy，您应该使用与该 Python 安装关联的 pip。以下是在该环境中安装 empy 版本 3.3.4 的命令：
> 
> ```
> ~/microros_ws/firmware/toolchain/espressif/python_env/idf4.1_py3.8_env/bin/pip3 install empy==3.3.4
> ```
> 
> 2. 之后，验证安装：
> 
> ```
> ~/microros_ws/firmware/toolchain/espressif/python_env/idf4.1_py3.8_env/bin/python3 -c "import em; print(em.__file__)"
> ```
> 
> 3. 更新 setuptools 和 wheel：
> 
> ```
> ~/microros_ws/firmware/toolchain/espressif/python_env/idf4.1_py3.8_env/bin/pip3 install --upgrade setuptools wheel
> ```
> 
> 5. 清理你的微型 ROS 工作区：
> 
> ```shell
> cd ~/microros_ws
> rm -rf build install log
> ```
> 
> 6. 重建工作区：
> 
> ```shell
> colcon build --symlink-install
> ```
> 
> 7. 获取安装文件：
> 
> ```shell
> source install/local_setup.bash
> ```
> 
> 8. 尝试再次构建固件：
> 
> ```shell
> ros2 run micro_ros_setup build_firmware.sh
> ```

It didn't work for me.