micro-ROS/micro-ROS-Agent #234: Deserializing error for nested message types

hr-ti, 2024-08-23:
**Describe the bug**
Nested message types in service types cause microROS on the MCU side to fail in the executor due to a deserializing error.

Service calls with non-nested message types work as expected. I've verified it's not a buffer overflow since the buffer size >> bytes received. There is also no packet loss in UDP traffic, the dest packet and src packet match 1:1. 

**To Reproduce**
Steps to reproduce the behaviour:
1. On MCU, init service with service type that has a nested message
2. On host side, call service through agent

**Expected behaviour**
The MCU accepts the service call and handles accordingly.

**Actual behaviour**
Executor fails due to deserialization error.

**System information (please complete the following information):**
 - OS: Ubuntu 22.04
 - ROS 2 Humble
 - MCU: STM32F767ZI

hr-ti, 2024-08-23:
Seems like this is an error with the nested message type's string member. As it fails in ucdr_deserialize_sequence_header, the length of the string member is greater than the capacity in rosidl_runtime_c__String, which is always 0. 

Since there's no issue with UDP comms, seems this is due an error in generating the service request from the agent side.

hr-ti, 2024-08-26:
This actually applies to any sequence type, like a byte array. Since capacity is always 0, any populated sequence simply just fails

gavanderhoorn, 2024-08-27:
Are you setting `capacity` to an appropriate value?

It isn't set automatically, so if you `bzero(..)` a msg struct or initialise it to `0`, 'the other side' will not see any elements in sequences, as `capacity` is not updated automatically.