micro-ROS/micro-ROS-Agent #240: Sudden humble build failures due to fastcdr changes

Ryanf55, 2024-11-25:
**Describe the bug**

I can no longer build. This has broken ArduPilot's CI, impacting our entire development team's ability to merge PR's.

**To Reproduce**
Steps to reproduce the behaviour:
1. Clone this repo with the `humble` branch in a workspace
2. Run `colcon build`
4. Observe the output

```
ryan@B650-970:~/Dev/ardu_ws$ colcon build --packages-up-to micro_ros_agent
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                             
Cloning into 'xrceagent'...
Switched to a new branch 'ros2'
CMake Error at CMakeLists.txt:153 (find_package):
  By not providing "Findfastcdr.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "fastcdr", but
  CMake did not find one.

  Could not find a package configuration file provided by "fastcdr"
  (requested version 2) with any of the following names:

    fastcdrConfig.cmake
    fastcdr-config.cmake

  Add the installation prefix of "fastcdr" to CMAKE_PREFIX_PATH or set
  "fastcdr_DIR" to a directory containing one of the above files.  If
  "fastcdr" provides a separate development package or SDK, be sure it has
  been installed.

gmake[5]: *** [CMakeFiles/uagent.dir/build.make:93: uagent-prefix/src/uagent-stamp/uagent-configure] Error 1
gmake[4]: *** [CMakeFiles/Makefile2:83: CMakeFiles/uagent.dir/all] Error 2
gmake[3]: *** [Makefile:91: all] Error 2
gmake[2]: *** [CMakeFiles/xrceagent.dir/build.make:86: agent/src/xrceagent-stamp/xrceagent-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/xrceagent.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
---
Failed   <<< micro_ros_agent [3.18s, exited with code 2]

Summary: 0 packages finished [3.90s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent

```

**Expected behaviour**

A stable `humble` branch won't have ABI breaks under our feet.

**System information (please complete the following information):**
 - OS: Ubuntu 22
 - ROS 2: ROS 2 humble
 - Version 1cb265e752f73003a37cb66ad849f6ef1d9ce695, tag `3.0.6`, branch `humble`

**Additional context**
https://github.com/ArduPilot/ardupilot/actions/runs/12014292947/job/33489746378?pr=28734#step:11:24

pablogs9, 2024-11-25:
Please @Ryanf55 test it now, I accidentally move a tag on Micro XRCE-DDS Agent

pablogs9, 2024-11-25:
Just for ensuring this not happens again: https://github.com/micro-ROS/micro-ROS-Agent/pull/241

closed 2024-11-25 by Ryanf55

pablogs9, 2024-11-25:
Thanks, @Ryanf55, sorry for the inconvenience.

Ryanf55, 2024-11-25:
No worries, thanks for resolving today.