micro-ROS/micro-ROS-Agent #243: micro_ros works on rmw_fastrtps_cpp but doesn't work with rmw_zenoh - should it?

PaulBouchier, 2025-02-11:
## System configuration

- Hardware description: Linux PC Ubuntu 24.04
- RTOS: Arduino environment running on ESP32 with serial link to PC
- Installation type: jazzy on ubuntu 24.04 PC. rmw_zenoh installed and can be activated by setting RMW_IMPLEMENTATION env var and starting zenoh router
- Version or commit hash: jazzy

#### Steps to reproduce the issue
1. Install an Arduino talker on esp32
2. Start micro_ros node with ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600
3. Run ros2 node/topic list. Observe the node and topic are published. ALL GOOD SO FAR
4. Shut down micro_ros_agent
5. Run 'pkill -9 -f ros && ros2 daemon stop' to top the fastrtps ros2 daemon per rmw_zenoh instructions
6. export RMW_IMPLEMENTATION=rmw_zenoh_cpp
7. start zenoh router in a terminal: ros2 run rmw_zenoh_cpp rmw_zenohd
8. Same as step 2: Start micro_ros node with ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600
9. Same as step 3: run ros2 node/topic list and no node or topic is shown from a terminal using the zenoh router. I also tried unsetting RMW_IMPLEMENTATION in another terminal to see if the micro_ros_agent was talking over fastrtps but it doesn't appear to

#### Expected behavior
I'm trying to make micro_ros work in a zenoh environment because my wifi just will not pass multicast (eero 6 mesh routers) so DDS just doesn't work at all - I can't get two hosts to exchange ROS messages.

micro_ros_agent should work seamlessly in either a rmw_fastrtps or a rmw_zenoh environment

#### Actual behavior
micro_ros_agent works in rmw_fastrtps environment but not in rmw_zenoh environment

#### Additional information
Arduino environment is described here: https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki#install-the-software-tools and the specific program installed on the esp32 is micro_ros_arduino_examples_platformio/micro-ros_reconnection_example, which creates a microros node: /int32_publisher_rclc which publishes an incrementing int on topic /std_msgs_msg_Int32.

When I start micro_ros_agent in the zenoh environment it prints out the normal "creating publisher" etc info, same as when it starts in the rmw_fastrtps_cpp environment:
`
bouchier@xps15:~$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600
[1739251929.823302] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 18
[1739251929.823603] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1739251929.999870] info     | Root.cpp           | create_client            | create                 | client_key: 0x6B9899EE, session_id: 0x81
[1739251929.999897] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x6B9899EE, address: 0
[1739251930.014532] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x6B9899EE, participant_id: 0x000(1)
[1739251930.019652] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x6B9899EE, topic_id: 0x000(2), participant_id: 0x000(1)
[1739251930.023036] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x6B9899EE, publisher_id: 0x000(3), participant_id: 0x000(1)
[1739251930.026887] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x6B9899EE, datawriter_id: 0x000(5), publisher_id: 0x000(3)
`

and ps shows the expected processes running, including a ros2 daemon using rmw_zenohd: 
`
$ ps -ef | grep ros
bouchier  103110  100488  0 23:31 pts/0    00:00:00 /usr/bin/python3 /opt/ros/jazzy/bin/ros2 run rmw_zenoh_cpp rmw_zenohd
bouchier  103113  103110  0 23:31 pts/0    00:00:00 /opt/ros/jazzy/lib/rmw_zenoh_cpp/rmw_zenohd
bouchier  103117  100620  0 23:32 pts/2    00:00:00 /usr/bin/python3 /opt/ros/jazzy/bin/ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600
bouchier  103120  103117  0 23:32 pts/2    00:00:00 /home/bouchier/ros2_ws/install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600
bouchier  103163   97772  7 23:33 pts/4    00:00:00 /usr/bin/python3 -c from ros2cli.daemon.daemonize import main; main() --name ros2-daemon --ros-domain-id 0 --rmw-implementation rmw_zenoh_cpp

`
ros2 node list shows no nodes running (with fastrtps it shows one node running), and ros2 topic list shows the default topics but not the one the micro_ros_agent is supposed to be publishing:
`
bouchier@xps15:micro-ROS-Agent$ ros2 node list
bouchier@xps15:micro-ROS-Agent$ ros2 topic list
/parameter_events
/rosout
`

I don't know whether this issue belongs in the micro_ros_agent tracker or the rmw_zenoh tracker. Please advise if I should close it here & open it in rmw_zenoh.

gavanderhoorn, 2025-02-11:
> micro_ros_agent should work seamlessly in either a rmw_fastrtps or a rmw_zenoh environment

as much as I would like this to be true, it isn't. This doesn't even work with two DDS-based RMWs (although it could, see ros2/rmw_cyclonedds#184 fi).

I'm curious to know what made you assume it would/should?

gavanderhoorn, 2025-02-11:
To clarify, I'm not a maintainer here, neither am I directly involved in the development of the Agent or micro-ROS. I just happen to be a user that ran into similar issues in the past (see the linked ros2/rmw_cyclonedds#184 fi).

> rmw_zenoh is billed as being a plug-in replacement for dds

IIUC, it's slightly more nuanced. RMWs are all expected to offer similar levels/kinds of functionality (ie: they must all support pub-sub, services and actions, parameters, etc). There is no requirement -- nor expectation really -- for all of them to be compatible, at the binary (ie: 'wire') level. That would also be impossible, given the wide range of technologies that can be used to implement an RMW.

So `rmw_zenoh`, as one RMW implementation, can indeed be used as a 'drop-in replacement' for a DDS-based RMW (ie: another RMW implementation), but that basically only means `rmw_zenoh` supports the same type of functionality, at a conceptual level. And concretely of course, `rmw_zenoh` implements the interface as defined for RMWs at the C function level, and is ABI compatible with all ROS 2 packages that have been compiled against that RMW C interface.

> I hoped that the micro_ros_agent would be rmw-agnostic.

I expect @pablogs9 (or someone else from eProsima) will comment on this and other parts of your comment.

pablogs9, 2025-02-11:
Hi @PaulBouchier and thanks for your clarifications @gavanderhoorn.

Indeed, DDS and Zenoh are not interoperable at the wire level, and probably they are not going to be.

We develop DDS solutions for ROS 2 and micro-ROS as DDS is now the default middleware solution for ROS 2. 
We have seen lots of RMW implementations (even [email RMW](https://github.com/christophebedard/rmw_email)) and they are not compatible with DDS out of the box. Sometimes there are bridges, sometimes not.

So in general do not expect Zenoh middleware to be compatible with the official releases of micro-ROS in the short term.

BR

closed 2025-02-11 by pablogs9

gavanderhoorn, 2025-02-11:
> Sometimes there are bridges, sometimes not.

@PaulBouchier: you could perhaps take a look at [eclipse-zenoh/zenoh-plugin-ros2dds](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds).

It's going to add one more 'hop' to your communication 'channels' though.