micro-ROS/micro-ROS-Agent #246: Micro-ROS agent Bluetooth

kartman88, 2025-03-18:
Is it possible to use Bluetooth as transport protocol for Micro-ROS agent or it is only meant for Serial and UDP?

lprobsth, 2025-03-19:
In theory there is no argument against using bluetooth for transort for XRCE.
See https://micro.ros.org/docs/tutorials/advanced/create_custom_transports/

You would have to implement the custom transport in the agent.

Geibinger, 2026-01-11:
I implemented a BLE transport for micro-ROS (both client and agent side). But it is not quite ready to share publicly. But if there is enough interest, let me know. I use a NUS (Nordic Uart Service) style communication with a few BLE parameters tweaked. On an ESP32c6, this is the throughput I accieve:

[image]

For now as I said, everything is within a private repo, but I'd be happy to share if it helps someone! :)